TUM-LIS/glip #23: Auto-detect UART

wallento, 2016-07-09:
We should start the UART transfers with a training sequence and auto-detect the baud speed with it.

imphil, 2016-08-17:
@wallento that's done, isn't it?

imphil, 2016-09-01:
currently auto-detect seems to be broken, connecting with speed=12000000 works for our standard OpTiMSoC design, without the speed option I get this:

~~~
$ opensocdebugd uart device=/dev/ttyUSB1
Open SoC Debug Daemon
Backend: uart
glip: gb_uart_open: Given speed 115200 did not work with device. Try autodetect..
glip: gb_uart_open: Could not autodetect baud rate.
glip: glip_open: Cannot open backend
glip: glip_write_b: No connection; you need to call glip_open() first!
glip: glip_read_b: No connection; you need to call glip_open() first!
opensocdebugd: /home/no56hud/src/optimsoc/external/opensocdebug/software/src/connection-standalone.c:67: receiver_thread_function: Assertion `rv == 0' failed.
Aborted
~~~