TUM-LIS/glip #30: Use FIFO implementation which doesn't cause warnings in Vivado

imphil, 2017-01-25:
Currently we use the oh sync fifos, which use memories which are inferred to block rams by vivado. However, it seems that it doesn't do this very nicely, given the DRC warnings we get in Ultrascale:

```
WARNING: [DRC 23-20] Rule violation (REQP-1902) RAMB36E2_AB_cascade_out_must_use_parity - The RAMB36E2 cell u_glip/u_out_buffer/mem/memory_dp/ram_reg_bram_9 has CASCADE_ORDER_B attribute set to MIDDLE with the CASDOUTB[31:0] bus used, but is missing connection(s) for the CASDOUTPB[3:0] bus.
```
(and similar).

I think I've also seen warnings earlier in the process for the 7series devices. We should look into that and use more FPGA-friendly FIFOs to reduce the warning-spew during the build.

imphil, 2017-01-25:
(Vivado 2016.3 that is)

wallento, 2017-01-26:
Yeah, those warnings are usually the reason we move back to inferring specific RAMs for some target devices.

I think the best approach is to try a few variations of writing RAM in Verilog and check which one works best. Otherwise I would expect this to repeat with another FIFO implementation again..

imphil, 2017-04-20:
@koenenwmn As discussed, have a look at the Xilinx RAM guidelines:

https://www.xilinx.com/support/documentation/sw_manuals/xilinx2016_4/ug901-vivado-synthesis.pdf
pg. 105

imphil, 2017-04-20:
The FIFO should then also be used for OpTiMSoC, see https://github.com/optimsoc/sources/issues/29

imphil, 2017-05-04:
Fixed with new FIFOs

closed 2017-05-04 by imphil