TUM-LIS/glip #32: Improve timing of glip_upscale

imphil, 2017-01-27:
glip_upscale shows up in timing reports. We should probably delay the output for one cycle to get better timing.

@wallento sounds reasonable?

imphil, 2017-01-27:
by "shows up in timing reports" I mean it's having a timing violation at 50 MHz :)

wallento, 2017-01-27:
Its actually not upscale, I think. Its where you put it. There should be registers as near to it as possible.

Or is it the state of upscale (its very simple IIRC)?

wallento, 2017-01-27:
But agree, the ingress path needs to be checked then.

imphil, 2017-01-27:
I'll have a look. It seems our default demo designs use WIDTH == 8, so these modules aren't really used in testing.

wallento, 2017-01-27:
Yeah, but then the MUX should hopefully be removed anyways. And that what it is a clog2(WIDTH)-Mux.

commit c616785f (2017-02-06): UART: Change module ordering in ingress data path

The main motiviation of this change is to improve the timing of the
ingress path. We move the ingress buffer directly to the user logic as
last element in the chain, the place where the upscaler has been sitting
before. This ensures that the fifo_in_* signals are a clean FIFO
interface with minimal combinatorial logic, thus users have more freedom
to connect their own logic on this interface.

To make such changes easier in the future, the code was refactored in
the following way:
- In module instantiations only wires named after the ports are [... cut]
closed 2017-02-06 by imphil