TUM-LIS/glip #36: Loopback test with FX3 backend

koenenwmn, 2017-04-20:
There is currently an issue when performing the loopback test with a multiple of 1024 bytes.
For small numbers (< 1024 * 24) nothing gets sent at all. For larger numbers the last 1024 bytes simply don't get sent. In both cases the test stalls forever. When sending 2 bytes more or less (or any other even number of bytes) everything works fine though.

I also noticed two exceptions to that: the test finishes normal if 1024 * 6, or a multiple of 1024 * 24 bytes, are sent.

imphil, 2017-05-04:
I've done the following debugging to narrow down the problem:

* Verified that the data gets sent to the device from the host: Using wireshark I see that 1024 bytes of data get sent to the device, and that the sending is acknowledged by the device.
* Verified that the data doesn't reach the hardware: using ILA I see that the FX3 never lowers its fx3_out_empty or fx3_out_almost_empty signals.

I strongly assume that the FX3 firmware is at fault here. It seems that it doesn't trigger a new "send data from internal FIFOs out to FPGA" action. 

@koenenwmn is this someting you can have a look at? Otherwise I'm sure we'll find a workaround on the host side to avoid this scenario by splitting up the transfer.

imphil, 2017-05-04:
I've found a rather reasonable fix in our software by sending a zero-length packet if our transmit buffers are empty. With that all transfers work fine.

commit a4b1ad37 (2017-05-04): FX3: Work around transfers not being received dev

Closes #36
closed 2017-05-04 by imphil