TUM-LIS/glip #39: FX3: Provide real 16 bit FIFO width

imphil, 2017-05-05:
We currently downscale the 32 bit FX3 interface to 16 bit at the FIFO side, with zero-padding if an odd number of 16 bit words is sent. This leads to padding zeros being present on the 16 bit FIFO side as data and we cannot distinguish it from real data.

Plan of attack:
- Provide two firmware images, one for 16 bit and one for 32 bit
- Make the FX3 FIFO width equal to the output FIFO width
- Remove all up and downscalers from the logic
- Optional: Add a control register to the firmware to give the host a chance to discover if it's 16 bit or 32 bit.
- Software: make sure that get_fifo_width() returns the correct value. If discovery can be implemented this is determined when connecting. Otherwise provide a backend configuration option (i.e. the user must know which device he/she is connecting to.)

koenenwmn, 2017-05-05:
The up-/downscalers are removed now and the parameter WIDTH determines the width of both the logic interface and the FX3 interface.
The FX3 firmware is now modified to write the FIFO width in number of Bytes to the serial number field of the USB descriptor, and get_fifo_width() now returns the correct value.
However, due to timing problems only the 32 bit interface is currently working.

koenenwmn, 2017-05-12:
Fixed in latest fx3 commit.

closed 2017-05-12 by koenenwmn