TUM-LIS/glip #41: cypressfx3 has different endian-ness than UART and TCP backends in 16 bit mode

imphil, 2017-05-17:
When integrating the cypressfx3 backend into OpTiMSoC I noticed an endianness issue with GLIP.
In the cypressfx3 backend with WIDTH set to 16 the bytes are swapped compared to the TCP and the UART backend. I'm not sure which backends show faulty behavior here (it could well be that TCP and UART are doing things wrong.)

The expected byte ordering is documented in http://www.glip.io/group__logicif.html 

We should check the uart, cypressfx3 and tcp backends to make sure they conform to the documentation.

imphil, 2017-05-17:
@koenenwmn is this something you could have a look at?

koenenwmn, 2017-05-17:
I set the byte order in the FX3 firmware to Big Endian. The problem should be solved now.

imphil, 2017-05-17:
Fixed in 02b668602197d22d13752348ade232adaf5576ac

closed 2017-05-17 by imphil