TUM-LIS/glip #45: FX3: Data corruption on heavy FPGA -> Host traffic

imphil, 2017-08-16:
In optimsoc we're seeing data corruption with the fx3 backend when transferring large amounts of data from the fpga to the host. At that time, no communication host -> FPGA happens. Also, the other direction (host->fpga) doesn't show the same problems (initializing the memory of the optimsoc system works without trouble, which equals heavy traffic host -> fpga with only some acknowledgement packets going from fpga to host).

Plan of attack:
1. Try to reproduce with GLIP only https://github.com/TUM-LIS/glip/issues/44
2. If not possible, try to get more insight into the GLIP FX3 logic backend while running inside OpTiMSoC (tricky, as the problem is rarely reproducible)

imphil, 2017-08-16:
Observations:
- The problem doesn't happen at the boundary of a USB transfer, but in the middle of it. This suggests that FX3 also didn't issue any backpressure signals as the internal buffers are not full.
- 4 bytes of data are inserted (at least in our testcase). These 4 bytes seem to be old data.
- No data is dropped, just 4 bytes inserted.

So the first guess would be a valid signal which is high too long.

imphil, 2017-08-16:
Suggestions are always wrong, so... ILA shows the flaga and flagb are low at the time when the transfer fails, i.e. the fx3 indicates its buffer is full.

imphil, 2017-08-16:
VCD file: https://gigamove.rz.rwth-aachen.de/d/id/v4CseqrEESdb23?11&id=v4CseqrEESdb23

The packet being sent out until clock 2910 is received fully and padded with two additional 2858, i.e. the resulting received data is:

~~~
0900
0000
0480
44e3
2858
e0dc
2cc0
dcdf
2cc0
0800
2858 <-- wrong!
2858 <-- wrong!
0900
0000
0480
b1e3
2858
a8dd
2cc0
a0dd
2cc0
0400
~~~

imphil, 2017-08-16:
Actually `oe` is simply high for two cycles too much:

[image]

koenenwmn, 2017-08-17:
The write signal should be fine that way, according to the GPIFII specs. I think the actual issue is that the egress_cdc gets empty during the write flush states, which is not handled properly. I pushed a fix in the new branch.

PR #46 (2017-08-17): Fixed bug regarding empty egress_cdc fifo.
The bug occured when the egress_cdc fifo got empty while being in
STATE_WRITE_DRAIN_1 or STATE_WRITE_DRAIN_2 and led to 1-2 wrong/too many
words being sent to the FX3.

Fixes #45
closed 2017-08-17 by imphil