TUM-LIS/glip #62: I am having trouble using glip_read_b function with FX3

chunyen627, 2018-10-23:
I refer to the loopback_measure.c program and modify it to perform from PC to FPGA memory 1GB data write and read tests.
I tried to call glip_write_b and glip_read_b function, Both length of data are set to 1GB data size,
glip_write_b function can successfully write 1GB data to FPGA memory, but when executing glip_read_b and length of data set greater than 150Kbyte ，FX3 will not respond.
Is there a limit to reading back large files? How can I read back large files at once? 
Thank you.

imphil, 2018-10-29:
Thanks @chunyen627 for this report. I'm not sure I fully understand your problem, so let me summarize what I understood. Please correct me where I'm wrong.

- You're using the cypress_fx3 backend.
- A test with the unmodified loopback_measure program works.
- You essentially want to have a loopback test with a block size of 1 GB, i.e. a single write of 1 GB, followed by a single read of 1 GB. This requires a buffer of more than 1 GB on the FPGA side. Did you modify the hardware demo code for that?

Questions:
- Are you using the 32bit or the 16 bit firmware?
- Which options are you using with the unmodified `loopback_measure`, i.e. what's the exact command  line you're using?
- Can you post the modified source code you're having trouble with?
- Does the non-blocking read work? (`glip_read()`)

chunyen627, 2018-10-30:
FX3 firmware :  SlaveFifoSync32.img
FPGA :  Xilinx Ultra SCALE ZCU102 with 2GB ddr4 (The 1GB data is pre-write to the DDR4 memory)

I have modified the hardware demo code,modified glip logic and added command decoder circuit.
First I will transfer 20byte command data from PC to FPGA glip logic,
the command decoder is responsible for reading data from the FPGA ddr4 and returning it to fx3.
Then call glip_read_b function reading FX3 from the PC, but if the read_block_size setting is greater than 150 bytes, the data can only be partially returned.

rv = glip_read_b(glip_ctx, 0, read_block_size, read_data, &size_read,0);
------------------------------------------------------------------------------------------------------------------
I have to track the util.c code and see that the program will stop at rv = cbuf_wait_for_level_change(buf, level);
It seems that the level value of level = cbuf_fill_level(buf) is always 0, even if it is not blocked, it does not work (glip_read()),

Currently using the solution for reading back large files, set read_block_size to 128 bytes as a unit, read back and merge data with loops.

imphil, 2018-11-02:
Ah ok, so my understanding was not fully correct. You send (host -> device) a command of 20 byte, and then the FPGA sends 1 GB of data. So the problem is only the receive path on the host?

As a first debugging step, could you set a breakpoint at in the receive loop of `usb_read_thread()` (unside `util.c`), e.g. in https://github.com/TUM-LIS/glip/blob/master/src/backend_cypressfx3/sw/backend_cypressfx3.c#L882 and step through the code there? Can you see that the `libusb_bulk_transfer()` function returns, and what amount of data is provided? Look around those functions a bit more to understand what is going on (adding `printf()`s should work as well).