TUM-LIS/glip #63: fx3_flagc_n and fx3_flagd_n of glip_cypressfx3_toplevel

eiselekd, 2019-03-23:
I was wondering about the flagc and flagd inputs in the FX3 toplevel entry: The CYUSB301X has only flaga and flagb described in its CYUSB3014 manual. Which pins are these?

koenenwmn, 2019-03-23:
That manual only gives a very rough overview of the GPIF II and the configuration of the FIFO interface. For a more detailed description please have a look at the Applicatino Notes AN65974. These notes also provide some design examples and describe the different flags.  
For our toplevel we use flags a and c as full and empty flags respectively, and flags b and d as partial full/empty flags.

eiselekd, 2019-03-30:
Can you also checkin the GPIF II project definition file?

closed 2019-04-09 by eiselekd