TUM-LIS/glip #64: Any anyone attempted a windows build?

AlessandroAU, 2019-03-29:
I got successful build on Linux, but I would really like windows binaries so I can use GLIP with LabView. 

I tried building with Cygwin and also cross-compling with mingw32 in Linux but I always end up getting stuck at some include file or unhelpful errors.

Has anyone done this? It is possible?

Regards,
Alessandro

imphil, 2019-03-29:
No, I don't know of anybody who used GLIP on Windows. Your success will also depend on the backend you're trying to use. 

First of all, don't cross-compile. A next step might be using something like msys2 to simplify the transition by having POSIX APIs available. You could also try using the new WSL on Windows 10. Beyond that I'd expect some Linux-only features to be used in GLIP which might not be available in a pure-POSIX environment. I'd hope that this would be a trivial code change in most places, but you'd never know ...

I'd encourage you to give it a try and see how far you get. Please report any bugs or errors you find and we might be able to help you out finding a good solution for particular issues.

AlessandroAU, 2019-03-29:
Much the same results with msys2. 

At least the following are missing in windows

syslog.h
netdb.h
sys/wait.h

imphil, 2019-03-29:
You'll need to work through these one by one to see what functions from that header are actually used, and how you could replace them with cross-platform code.